nordicsemi/Android-nRF-Connect #204: Is McuMgr with DFU not supported with automated testing

git-vcs, 2023-11-21:
**Describe the bug**
I am trying to make an automated firmware upgrade test. I followed the examples supplied in the documentation but I am getting an error.

Here is the error i am getting:
V    14:29:13.718    [DFU] DFU service started
V    14:29:13.719    [DFU] Opening file...
I    14:29:13.737    [DFU] Firmware file opened successfully
V    14:29:13.737    [DFU] Connecting to DFU target...
D    14:29:13.738    [DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M)
I    14:29:13.752    [DFU] Connected to 64:51:54:90:8C:A5
D    14:29:13.755    [DFU] wait(1600)
V    14:29:15.357    [DFU] Discovering services...
D    14:29:15.358    [DFU] gatt.discoverServices()
I    14:29:15.375    [DFU] Services discovered
W    14:29:15.377    [DFU] DFU Service not found
V    14:29:15.377    [DFU] Disconnecting...
D    14:29:15.412    [DFU] gatt.disconnect()
I    14:29:15.412    [DFU] Disconnected
D    14:29:15.412    [DFU] gatt.disconnect()
D    14:29:15.416    [DFU] gatt.close()
D    14:29:15.426    [DFU] wait(600)
V    14:29:16.033    Cleaning up...
V    14:29:18.051    Disconnecting...
D    14:29:18.051    gatt.disconnect()
D    14:29:18.076    [Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I    14:29:18.076    Disconnected
D    14:29:18.076    gatt.close()
D    14:29:18.092    wait(200)

When I try to use the DFU button in the app I see the logs mention a “McuMgr”.
I therefore wonder if there is a way to make the automated test use McuMgr?

The issue is only in the automated test, the dfu button in the app works great.

philips77, 2023-11-23:
Hi, sorry for that. Indeed, that feature is still in our backlog. I'll mark the issue as a feature request.

closed 2024-01-16 by philips77

philips77, 2024-01-16:
Please, use version 4.28.0 and the new **mcu** operation.

PR #209 (2024-01-16): MCU Manager support in Automated tests
This PR fixes #204.