nordicsemi/Android-nRF-Connect #205: How to advertise a service with multiple characterisitcs?

immrn, 2023-11-25:
I'm using the app as an advertiser.
But I can't find an option to add a second characteristic to my existing service.

https://github.com/NordicSemiconductor/Android-nRF-Connect/assets/49674379/7909dd67-2ed7-4506-8c1c-895c7bce5bee

Is the app capable of creating an advertised service with multiple characteristics?
And how can I achieve it?

philips77, 2023-11-26:
Hi,
The ADVERTISER tab defines advertising packets. This is what the phone may advertise (broadcast) with. A device may claim support for any Service UUID, may contain some manufacturer data, service data, etc. Those go into the advertising packet or scan response data.

This is not what a connected device would discover after connecting to your phone. If you mark the advertising packet as connectable, any remote device may connect to your phone. It will discover all services added by any of the apps running on the phone. In nRF Connect for Mobile you add services, characteristics and descriptors using hamburger menu -> Configure GATT Server. There, create a new configuration using the top menu on the middle, import one, or use a predefined one. The configuration you set up will be visible for a connected device, and it will be able to send/read data.

closed 2023-11-26 by philips77

immrn, 2023-11-26:
Thanks a lot!