nordicsemi/Android-nRF-Connect #207: 4.27.0 app crashes during automated tests

utisam, 2024-01-10:
**Describe the bug**
After upgrading to 4.27.0, `org.simpleframework.xml.core.ConstructorException` is raised during an automated test.

**To Reproduce**
Running automated tests.

**Expected behavior**
NA

**Screenshots**
NA

**Versions (please complete the following information):**
 - Android version: 12 / 14
 - Phone model: Pixel 3a (Android 12) / Pixel 5a (Android 14)
 - App Version: 4.27.0

**Additional context**

Stack trace says `No constructor accepts all read only values in class no.nordicsemi.android.mcp.test.domain.command.Connect` but the connect tag in our file is still `<connect timeout="5000" />`.

<details>
<summary>Stack trace from logcat</summary>

```
01-10 18:57:03.579  3915  4043 W TestService: Error while unmarshalling XML
01-10 18:57:03.579  3915  4043 W TestService: org.simpleframework.xml.core.ConstructorException: No constructor accepts all read only values in class no.nordicsemi.android.mcp.test.domain.command.Connect
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.InstantiatorBuilder.validateConstructors(SourceFile:15)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.InstantiatorBuilder.validateConstructors(SourceFile:7)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.InstantiatorBuilder.validate(SourceFile:58)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.InstantiatorBuilder.build(SourceFile:3)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.StructureBuilder.commit(SourceFile:7)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.ObjectScanner.validate(SourceFile:7)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.ObjectScanner.scan(SourceFile:10)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.ObjectScanner.<init>(SourceFile:22)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.ScannerFactory.getInstance(SourceFile:36)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Support.getScanner(SourceFile:3)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Source.getScanner(SourceFile:3)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Source.getSchema(SourceFile:1)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.read(SourceFile:16)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.read(SourceFile:7)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Traverser.read(SourceFile:2)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.CompositeInlineList.read(SourceFile:13)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.CompositeInlineList.read(SourceFile:10)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.CompositeInlineList.read(SourceFile:3)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.CompositeListUnion.readElement(SourceFile:5)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.CompositeListUnion.read(SourceFile:2)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readVariable(SourceFile:47)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readInstance(SourceFile:1)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readUnion(SourceFile:1)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readElement(SourceFile:76)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readElements(SourceFile:25)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.access$400(SourceFile:1)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite$Injector.read(SourceFile:27)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.read(SourceFile:19)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.read(SourceFile:7)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Traverser.read(SourceFile:2)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.CompositeInlineList.read(SourceFile:13)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.CompositeInlineList.read(SourceFile:10)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.CompositeInlineList.read(SourceFile:3)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readVariable(SourceFile:47)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readInstance(SourceFile:1)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readUnion(SourceFile:1)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readElement(SourceFile:76)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.readElements(SourceFile:25)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.access$400(SourceFile:1)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite$Injector.read(SourceFile:27)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.read(SourceFile:19)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Composite.read(SourceFile:7)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Traverser.read(SourceFile:2)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Persister.read(SourceFile:16)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Persister.read(SourceFile:15)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Persister.read(SourceFile:13)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Persister.read(SourceFile:10)
01-10 18:57:03.579  3915  4043 W TestService:   at org.simpleframework.xml.core.Persister.read(SourceFile:3)
01-10 18:57:03.579  3915  4043 W TestService:   at no.nordicsemi.android.mcp.test.TestService.onHandleIntent(SourceFile:304)
01-10 18:57:03.579  3915  4043 W TestService:   at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:77)
01-10 18:57:03.579  3915  4043 W TestService:   at android.os.Handler.dispatchMessage(Handler.java:106)
01-10 18:57:03.579  3915  4043 W TestService:   at android.os.Looper.loopOnce(Looper.java:205)
01-10 18:57:03.579  3915  4043 W TestService:   at android.os.Looper.loop(Looper.java:294)
01-10 18:57:03.579  3915  4043 W TestService:   at android.os.HandlerThread.run(HandlerThread.java:67)
```

</details>

philips77, 2024-01-15:
Hi,
We released 4.27.1 last Friday with this issue fixed. Have you tried it?

philips77, 2024-01-15:
Also, we added a script *test.sh* for Mac / Unix.

closed 2024-01-15 by philips77

utisam, 2024-01-18:
Thank you. It works.