nordicsemi/Android-nRF-Connect #211: Unable to see "Indicate" notifications in log

Srikor, 2024-02-07:
Hello Team. We are in the process of developing remote patient monitoring capabilities by using various BLE devices. As part of it we have purchased multiple devices and one of them is a Rossmax Bluetooth enabled Thermometer. I installed the Android App provided by Rossmax and was able to see the temperature that was read by the BLE thermometer. 

I installed nRF connect and was trying to do similar thing. I was able to scan for the thermometer and enable indications to get notified when temperature is read by the device. While the logs indicate **"Indications enabled" sent**, I'm not able to see any logs when temperature measurement is done. Want to understand if nRF connect app is capable of getting indications and notifications and if so how can I see them in the app.

philips77, 2024-02-07:
Yes, they should be visible both in the list of services under the indicating characteristic and in the log.

Srikor, 2024-02-07:
Thanks for the response. I can see the Temperature service listed and I'm able to enable indications. The app does mention "Indications Enabled" under descriptor and in the device the bluetooth icon also becomes steady indicating successful pairing. Yet when I measure the temperature in the device, I'm not seeing the temperature in the log. The mobile runs on Android API 33 and is Samsung mobile.

philips77, 2024-02-07:
Perhaps the indication is somehow triggered from the device? Indications and notifications should be shown. Read responses are shown only when read was requested from nRF Connect, not some other app.

Srikor, 2024-02-08:
So, I wasn't getting any indication under the Temperature service when I measure the temperature. So, I enabled indications on all services that are marked "Indicate" or "Notify". And every time I read the temperature using the thermometer I received notifications in another unknown service. Part of the log is below.

A 14:21:31.098 "Notifications enabled" sent
V 14:21:31.106 Notifications enabled for 0000fff3-0000-1000-8000-00805f9b34fb
V 14:21:31.855 Enabling notifications for 0000fff4-0000-1000-8000-00805f9b34fb
D 14:21:31.855 gatt.setCharacteristicNotification(0000fff4-0000-1000-8000-00805f9b34fb, true)
D 14:21:31.859 gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x0100)
I 14:21:31.973 Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
A 14:21:31.974 "Notifications enabled" sent
V 14:21:31.980 Notifications enabled for 0000fff4-0000-1000-8000-00805f9b34fb
I 14:21:38.707 Notification received from 0000fff1-0000-1000-8000-00805f9b34fb, value: (0x) A5-0C-32-11-06-24-5C-51-6A-FA-E5-03-C9
A 14:21:38.707 "(0x) A5-0C-32-11-06-24-5C-51-6A-FA-E5-03-C9" received
I 14:21:39.295 Notification received from 0000fff1-0000-1000-8000-00805f9b34fb, value: (0x) A5-0C-32-11-06-24-5C-51-6A-FA-E5-03-C9
A 14:21:39.295 "(0x) A5-0C-32-11-06-24-5C-51-6A-FA-E5-03-C9" received
I 14:21:39.978 Notification received from 0000fff1-0000-1000-8000-00805f9b34fb, value: (0x) A5-0C-32-11-06-24-5C-51-6A-FA-E5-03-C9
A 14:21:39.978 "(0x) A5-0C-32-11-06-24-5C-51-6A-FA-E5-03-C9" received
I 14:21:40.562 Notification received from 0000fff1-0000-1000-8000-00805f9b34fb, value: (0x) A5-0C-32-11-06-24-5C-51-6A-FA-E5-03-C9
A 14:21:40.562 "(0x) A5-0C-32-11-06-24-5C-51-6A-FA-E5-03-C9" received
V 14:22:15.625 Reading characteristic 0000fff5-0000-1000-8000-00805f9b34fb
D 14:22:15.625 gatt.readCharacteristic(0000fff5-0000-1000-8000-00805f9b34fb)
I 14:22:15.752 Read Response received from 0000fff5-0000-1000-8000-00805f9b34fb, value: (0x) 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
A 14:22:15.752 "(0x) 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00" received
D 14:23:31.921 [Callback] Connection state changed with status: 19 and new state: DISCONNECTED (0)
W 14:23:31.921 Connection terminated by peer (status 19)
I 14:23:31.921 Disconnected
D 14:23:31.986 [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED. 

Could it be that the manufacturer is sending notifications under different service. Does the value look like a temperature measurement?

philips77, 2024-02-08:
Seems like the manufacturer decided to use proprietary service for temperature measurements, not a standard one. That means, that the value can be encoded in any way.

If you know what value was shown, you could try to reverse engineer the value from the bytes.

Srikor, 2024-02-08:
Yeah. I have the value. Will try to reverse engineer it. Thanks for your help!

closed 2024-02-08 by Srikor