nordicsemi/Android-nRF-Connect #222: can not show notify data more than 20 chars

skyyearxp, 2024-09-18:
**Describe the bug**
when send a request to mcu, and then mcu notify back 20 chars is ok, when 21 chars there is nothing in log view.

**To Reproduce**
Steps to reproduce the behavior:
1. connet my mcu device
2. enable notification
3. send request like "test"
4. if notify return is 20 chars, it will be show in log view
5. if notify return more than 20 chars, it will not be seen in log view

**Expected behavior**
all notify return chars will be seen in log view.

**Screenshots**
[image]

**Versions (please complete the following information):**
 - Android version: 11
 - Phone model SHARK KSR-A0
 - App Version  4.24.3/4.25.3/4.27.1/4.28.1

**Additional context**
NO

skyyearxp, 2024-09-18:
after google, i found i need to set  “Request MTU” to Larger Number will be work...

philips77, 2024-09-18:
You may also find a setting in *Settings / Connectivity / Auto request maximum MTU* so you don't have to do this every time manually.

closed 2024-09-18 by philips77