nordicsemi/Android-nRF-Connect #224: Test suite file not found in Test/<xxx>.xml

wencyxie, 2024-11-20:
Hi
I wrote a simple xml to read information from a device and it worked very well on my phone. I could get the information I want.

But, when I try to use another phone, still the same xml and the same device, it always stucks and nRF Logger reports "Test suite file not found in Test/<xxx>.xml".

I used "adb shell" to check the xml path, and I can find the xml file in the right path: /sdcard/Android/data/no.nordicsemi.android.mcp/files/Test/

I am very confused now. Every step is the same. Why it worked on some phones and couldn't work on another?
So I hope you could help me learn about:
1. What situation will report this log?
2. What can I do to make it work?

philips77, 2024-11-20:
By the file name, or a lack of it, I suspect it's something with env variables? The file name should not be empty, so it's being replaced with "" somewhere on the way. Do you see this file as `.xml`, or `somename.xml`?

Check your script and try to print the name sent, or enable echo.

wencyxie, 2024-12-16:
The name of the file is good. I omitted it before. The real log is "Test suite file not found in Test/test_read.xml". I echoed the name before "adb shell am start-...-service..." and it is also fine.
If there is something wrong with env variables, do you have any idea what env variables I should check or add?

wencyxie, 2024-12-16:
Hi I may just find the reason. It may be a permission problem. Here is the phone I can't use normally. It is with "u0_a360". And I check a phone I can use normally. It should be with "shell".
[image]

closed 2024-12-16 by wencyxie