nordicsemi/Android-nRF-Connect #225: Firmware transfer via dfu function of nRF-connect app version 4.29.0

Rutuja2906, 2024-12-05:
[image]
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'nRF-Connect App' for firmware update
2. Click on 'Connect' to connect the peripheral BLE device 
3. Then click on 'DFU'
4. Error as shown in the attached screenshot. The app is not showing the progress graph of the firmware transfer as it is not initiating, sometimes crashing the app and showing a message as shown the attachment.

**Expected behavior**
As per the old version, when the DFU option is selected, it should navigate to select firmware to be transferred and then firmware transfer should start with the progress graph on a mobile screen.

**Screenshots**
I have added a screenshot of the error getting while updating firmware.

**Versions (please complete the following information):**
 - Android version: [14]
 - Phone model [OnePlus Nord 2 and Samsung Galaxy S23]
 - App Version [4.29.0]

philips77, 2024-12-06:
Thank you for the bug report. We'll fix it as soon as possible.

philips77, 2024-12-11:
Hi,
I found an issue when a _.bin_ file is selected, but it doesn't crash the app.
Could you attach your file which you select here, or does the crash happen even before you have a chance to select a file?

Rutuja2906, 2024-12-11:
Hello,

Please refer to the attachment of file I used for dfu.

Best regards,
Rutuja

philips77, 2024-12-11:
I just released version 4.29.1. Please try it when it gets available in Play Store.