nordicsemi/Android-nRF-Connect #226: Missing a period of notification in the log file when comparing to wireshark capture

yoyuasazenth, 2024-12-20:
**Describe the bug**
I currently have a system which uses BLE5.0 to stream some raw sensor data via notification through service uuid 0xfff0 and characteristic uuid 0xfff7.
The transmission data rate is about 1.25 kB/s.
The issue i'm currently facing is that sometimes i found missing notification for a long period of time (11 seconds) when wireshark capture shows that the notification packet has been transmitted successfully.
It seems to me that the packet has been transmitted from peripheral device (sensor) to central device (mobile) however for some reason the notification data is not logged into the log file of nRF Connect application.
Therefore, i'd like to know if there is any condition in the application that could caused this behavior?

**Expected behavior**
The notification should not be missing from the log file if wireshark capture shows that the notification packet has been transmitted successfully.

**Screenshots**
[image]

**Versions (please complete the following information):**
 - Android version: 12
 - Phone model: Samsung Galaxy Tab S7 FE TE119
 - App Version: 4.29.1

philips77, 2024-12-30:
Hi,
It may be that the phone failed to scan those packets, e.g. the controller was busy with something else and missed packets. The app is logging all packet that it gets from the system.

yoyuasazenth, 2025-01-02:
> Hi, It may be that the phone failed to scan those packets, e.g. the controller was busy with something else and missed packets. The app is logging all packet that it gets from the system.

Hi,
I have done another test and it seems that the mobile device seems to received the notification packet normally but on the nRF Connect app side it does not received the packet.

This wireshark log is exported from the mobile device itself from btsnoop_hci.log file (this is why i think that mobile device receives the packet successfully since all the packets is present in the android hci log file).
[image]
From the above image, you can see that there is continuous notification from 18:18:38.461007 onwards.

However, from the nRF connect log during the same session, it can be seen that there is notification discontinuity from time 18:18:38.467 and then notification resumes at time 18:19:13.679
[image]

yoyuasazenth, 2025-01-03:
It seems like if i clear cache and clear app data of both nRF Connect and nRF Logger application and restart device then the issue is resolved.