nordicsemi/Android-nRF-Connect #230: Release source and F-Droid

rugk, 2025-01-04:
It would be great to see this Android app on F-Droid!

[F-Droid](https://f-droid.org/) is an Android app store specifically for free/libre open-source apps. It would be great if your app could be released there, as it is the number one for getting FLOSS Android apps for many people.
F-Droid also builds all apps from source ([optionally even reproducible](https://f-droid.org/en/docs/Reproducible_Builds/)), so downloads from there can be trusted.

Of course this would require you to open-source the app. But imho that would be a good thing actually, and for a free app I see no disadvantages. And you could feature the app and this the brand on the store too, as explained. 
At least one app could be there to hint to the other apps. 

The [app developer FAQ](https://f-droid.org/en/docs/FAQ_-_App_Developers/) or [the quick start guide](https://f-droid.org/en/docs/Submitting_to_F-Droid_Quick_Start_Guide/) may help you to get started.

BTW a release on F-Droid could also bring some (more) popularity (in case that is intended), as it will show up in the app (new apps are featured there).

Anonymous941, 2025-06-07:
Even the nRF Connect desktop app, which is essentially the same thing for a different platform, is open source

philips77, 2025-06-07:
Why don't you want to use it with closed source? Perhaps I can help you get your answers without releasing the source code?
I have to be honest, this app was mostly written aroudn 2013-2015 and the source is complex. If follows architecture patterns from a decade ago. I am more than happy to answers any of your questions, but I daubt the source code would help in any way.