nordicsemi/Android-nRF-Connect #231: nRF connect mobile app stucks in bonding state

Shreeyash-17, 2025-01-28:
**Description** :

I’m encountering an issue when connecting to a peripheral device from a mobile device nRF Connect app. During the bonding process, the console log on the peripheral confirms that all three phases of Bluetooth Low Energy pairing are completed successfully, and keys are distributed correctly.

However, in the nRF Connect app, the process remains stuck in the bonding state, which is unexpected. The OTA log also clearly shows that key distribution has occurred as expected.

[image]

**To Reproduce**:

1. Use any peripheral device for Ble pairing.
2. While performing this process extended advertising is enabled on peripheral device .
3. Connect from nRF Connect app to peripheral  device.
4. Perform bonding with the peripheral from a mobile device using nRF Connect app.
5. Observe the peripheral’s console log  and the nRF Connect app during bonding.

**Observed Behavior**:

- The peripheral console confirms successful completion of pairing and key distribution.
- nRF Connect remains stuck in the bonding state.

 
**Expected behavior**
Once key distribution is completed, the bonding process should complete successfully in the nRF Connect app as well.

**Screenshots** 
 
[image]

**Versions :**
 - Android version: 14,11,10,13
 - Phone model(Android version) :  OnePlus 11R 5G(14),Redmi note 11 (13),oppo A15(10),Vivo 1920(11),One plus Nor CE 5G(11),RealME C15(11),Poco M6.
 - App Version : v4.29.1

**nRF Connect Logs**:
Below is the log captured during the issue:

nRF Connect, 2025-01-27
spoiu1234 (F4:12:FA:E3:A6:DA)
V 16:03:44.684 Connecting to F4:12:FA:E3:A6:DA...
D 16:03:44.684 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D 16:03:45.776 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 16:03:45.776 Connected to F4:12:FA:E3:A6:DA
V 16:03:45.777 Discovering services...
D 16:03:45.777 gatt.discoverServices()
D 16:03:46.009 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I 16:03:46.036 Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D 16:03:46.166 [Callback] Services discovered with status: 0
I 16:03:46.167 Services discovered
V 16:03:46.172 Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Server Supported Features [R] (0x2B3A)
- Client Supported Features [R W] (0x2B29)
D 16:03:46.172 gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
I 16:03:46.254 Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 5000ms)
I 16:03:46.704 PHY updated (TX: LE 2M, RX: LE 2M)
V 16:03:48.999 Starting pairing...
D 16:03:48.999 device.createBond()
D 16:03:49.039 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BONDING (11)
D 16:03:49.502 [Broadcast] Action received: android.bluetooth.device.extra.PAIRING_VARIANT, pairing variant: CONSENT
V 16:05:07.737 Disconnecting...
D 16:05:07.737 gatt.disconnect()
D 16:05:07.793 [Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I 16:05:07.793 Disconnected

philips77, 2025-01-29:
Hello,
nRF Connect just shows states reported by Android system. The last broadcast shown in the logs says `bond state changed to: BONDING (11)`, that's why this value is displayed.
Aren't you getting "double" notification on Android for pairing? We often get first with PAIR button, which opens a dialog to accept pairing and then another notification, exactly the same, where we need to tap PAIR again and then again accept the paiting on a new dialog. Only after that the bond state changes to `BONDED`.

Shreeyash-17, 2025-01-30:
@philips77 Thanks for your response.

I am experiencing this issue only with some phones. Interestingly, two different phones with the same Android version show contradictory behavior, which is why I initially suspected that the issue might be on the app side.

On Android, I receive double notifications for pairing, and even after accepting both, the bonding process gets stuck and does not transition to the Bonded state. However, on other phones, the bonding completes successfully, and the state changes to Bonded.

From the logs, it is clear that the pairing has been completed successfully from the peripheral side, meaning it is now the phone's responsibility to store the exchanged bonding information.

If the phone fails to store the bond information, then there should not be an option to delete bond information. However, in this case, the option is available, which indicates that the bond information has been stored, but the app still shows the Bonding state instead of Bonded.

If the bond information is stored correctly, then the delete bond option should still be available, and the state in the app should correctly update to Bonded.

philips77, 2025-02-04:
> If the phone fails to store the bond information, then there should not be an option to delete bond information. However, in this case, the option is available, which indicates that the bond information has been stored, but the app still shows the Bonding state instead of Bonded.

I think in the app we just make _Delete bond information_ if the state is != BONDED. What happens when you try to delete it in this BONDING state?

Shreeyash-17, 2025-02-07:
@philips77 The connection gets disconnected if I try to delete it while in the BONDING state. Could this be an Android issue?